Hypfer/deye-microinverter-cloud-free #2: Hints for protocol decoding

abrain, 2023-08-03:
Hi, thanks for publishing the dummy cloud! I also worked on deciphering the protocol over the past weeks and we basically found the same things. I might have some additions, I'd need to check in detail.

Until now, I only had access to packets from the SUN300G3 and I see you are using the 600 model. There should be a value specifying the number of usable/used MPPT paths, but I see several bytes with the value `0x01`.

The relevant payload bytes are: 15, 131, 132, 228, 252
Is one of them constantly `0x02` for you?

Hypfer, 2023-08-04:
I'm not entirely sure which bytes you mean, but you can use [the kaitai struct](https://github.com/Hypfer/deye-microinverter-cloud-free/blob/master/assets/deye_sun600.ksy) at [ide.kaitai.io](https://ide.kaitai.io) to compare different payloads in an interactive dissection.

In the attached zip, you will find a data message from a sun600
[data_msg_sun600.zip](https://github.com/Hypfer/deye-microinverter-cloud-free/files/12260292/data_msg_sun600.zip)

abrain, 2023-08-23:
Awesome, thanks! I was able to identify the byte in question:
```js
numberOfMpptPaths: payload.readInt8(131)
```

Maybe these are useful to you as well:

```js
dailyYield: payload.readUInt32LE(33) / 100,
totalYield: payload.readUInt32LE(37) / 10,
startupSelfCheckingTime: payload.readUInt16BE(243),
previousUpdate: new Date(
    payload.readUInt8(245) + 2000,
    payload.readUInt8(246) + 1,
    payload.readUInt8(247),
    payload.readUInt8(248),
    payload.readUInt8(249),
    payload.readUInt8(250)
)
```

Hypfer, 2023-08-23:
Are you sure about that? Because for the dailyYield offset, I end up in the middle of the inverter SN

closed 2023-08-23 by Hypfer