Hypfer/deye-microinverter-cloud-free #9: decreasing "Grid energy today"

vordenken, 2023-10-20:
Hi Hypfer,
thanks for this awesome project. I'm running the dummy cloud and for some reason the "grid energy today" value seems to be decreasing? This also happens for the "Grid energy total" (see screenshots)

[image]
[image]

Hypfer, 2023-10-20:
Ah yes that's just winter in germany. The sun needs to be recharged for next summer.

But for a more serious answer:
https://github.com/Hypfer/deye-microinverter-cloud-free/blob/e35914b2d1f08cc1431ccd628f65427f390acc90/dummycloud/src/Protocol.js#L113-L117

The code just reads 8 bytes from the payload sent by the inverter. It doesn't do any sum calculations or keeping track of energy production over time.
Whatever it is that is happening there, it is happening inside the firmware of the inverter and/or its logger module.

Are you on a recent firmware?

vordenken, 2023-10-20:
That's what I thought would be the case first as well :p 

Ok interesting, thanks for clarifying. I just checked the firmware and it was on "MW3_16U_5406_1.57" so I upgrade to the latest one "MW3_16U_5406_2.32". I'll keep an eye on the values today/the next days and give an update :)

vordenken, 2023-10-25:
FYI: A few hours after upgrading the fw to 2.32, the device stopped working. It's till now at 0 watts generating and the radiator temperature reads "-10".
This seems to be a known problem -> https://www.borncity.com/blog/2023/09/09/deye-wechselrichter-sun600g3-eu-230-nach-anschluss-des-relaisbox-defekt
So the latest is not always the greatest :D