Hypfer/deye-microinverter-cloud-free #14: Receiving bullshit multiple data packets, many containing bullshit

czoller, 2024-04-08:
Hi, I'm running the Dummycloud with a Deye SUN-M80G3-EU-Q0  with Firmware MW3_16U_5406_2.32-D1 and I receive a lot of data packages that contain bullshit after being parsed. Most of the time, I receive multiple packages in row and often the first one contains reasonable data and the latter bullshit. But this is not always the case.

Here are some example logs:
```
 [2024-04-08T17:14:03.092Z] [TRACE] 2024-04-08T17:14:03.092Z Data received from client: �JB,��Tgi��e�2305107146Z�        ��
▒�▒�▒�▒�V0.2.0.1V0.1.1.7V1.2.1.8@'q2305107146-1    2305107146-2    2305107146-3    2305107146-4    �,
�&�<▒   ▒��,I
�&�=

[2024-04-08T17:14:03.092Z] [TRACE] 2024-04-08T17:14:03.092Z Data a54a01104215162c9ad6ea0108540516670069030000d60fad6501008e0500003233303531303731343604005a000000ef06000000000000100900000000000000000000921304000000d20a18fc18fc18fc18fc0000000000000000000000000f01000016010000000000000000000056302e322e302e3156302e312e312e3756312e322e312e38000000001f4002010004000400040000000000000327000003710000000000000000323330353130373134362d3120202020323330353130373134362d3220202020323330353130373134362d3320202020323330353130373134362d3420202020139c002c00010000000000000abe0726141e128e003c1804081109010001000102011712180000000000000000000000000000000000ff000000000000139c002c00000000000000000049000000000001000000000abe0726141e128e7f3d0000000000000000000000020c15
[2024-04-08T17:14:03.092Z] [DEBUG] Received packet of type "DATA"
[2024-04-08T17:14:03.092Z] [DEBUG] DATA packet data from ::ffff:192.168.179.10 {
  pv: {
    '1': { v: 27.1, i: 0, w: 0, kWh_today: 0.4, kWh_total: 80.7 },
    '2': { v: 27.8, i: 0, w: 0, kWh_today: 0.4, kWh_total: 88.1 },
    '3': { v: 0, i: 0, w: 0, kWh_today: 0, kWh_total: 0 },
    '4': { v: 0, i: 0, w: 0, kWh_today: 0, kWh_total: 0 }
  },
  grid: {
    active_power_w: 4,
    kWh_today: 0.9,
    kWh_total: 177.5,
    v: 232,
    i: 0,
    hz: 50.1
  },
  inverter: { radiator_temp_celsius: 27.7 },
  inverter_meta: {
    rated_power_w: 800,
    mppt_count: 2,
    startup_self_check_time: 60,
    current_time: 2024-04-08T17:09:01.000Z,
    grid_freq_hz_overfreq_load_reduction_starting_point: 50.2,
    grid_overfreq_load_reduction_percent: 44,
    grid_v_limit_upper: 275,
    grid_v_limit_lower: 183,
    grid_freq_hz_limit_upper: 51.5,
    grid_freq_hz_limit_lower: 47.5,
    protocol_ver: 'V0.2.0.1',
    dc_master_fw_ver: 'V0.1.1.7',
    ac_fw_ver: 'V1.2.1.8'
  }
}
[2024-04-08T17:14:03.093Z] [DEBUG] Data event emitted.
[2024-04-08T17:14:03.093Z] [TRACE] Response a50a00101216162c9ad6ea0101db251466000000005a15
[2024-04-08T17:14:03.297Z] [TRACE] 2024-04-08T17:14:03.297Z Data received from client: �*B,���Tgi��e�3806571563-01230510714�"o��
                                                                                                                                ;
                                                                                                                                 ;&

                                                                                                                                   ��

,

        ����A   �#x@�,�.�0�
                           ���`���@��LSW5BLE_MW17_5406_RELAY_270_1.00-SH#
[2024-04-08T17:14:03.298Z] [DEBUG] Received packet of type "DATA"
[2024-04-08T17:14:03.298Z] [DEBUG] DATA packet data from ::ffff:192.168.179.10 {
  pv: {
    '1': { v: 0, i: 0, w: 0, kWh_today: 1, kWh_total: 195.5 },
    '2': { v: 0, i: 3339.1, w: 0, kWh_today: 1, kWh_total: 475 },
    '3': { v: 474.3, i: 25.6, w: 12142.08, kWh_today: 1, kWh_total: 253 },
    '4': { v: 0, i: 0, w: 0, kWh_today: 231.4, kWh_total: 2 }
  },
  grid: {
    active_power_w: 0,
    kWh_today: 0.49,
    kWh_total: 89235128.2,
    v: 1387.6,
    i: 0.8,
    hz: 0
  },
  inverter: { radiator_temp_celsius: 0 },
  inverter_meta: {
    rated_power_w: 30,
    mppt_count: 0,
    startup_self_check_time: 0,
    current_time: Invalid Date,
    grid_freq_hz_overfreq_load_reduction_starting_point: 0,
    grid_overfreq_load_reduction_percent: 256,
    grid_v_limit_upper: 0,
    grid_v_limit_lower: 0,
    grid_freq_hz_limit_upper: 0,
    grid_freq_hz_limit_lower: 0,
    protocol_ver: '\x00\x02\x00\x00\x00\x02\x0B;',
    dc_master_fw_ver: '\x0B;\x07&\x04\x0B\x04\x0B',
    ac_fw_ver: '\x14\x1E\x14\x1E\x12\x0E\x12\x0E'
  }
}
[2024-04-08T17:14:03.299Z] [DEBUG] Data event emitted.
[2024-04-08T17:14:03.299Z] [TRACE] Response a50a00101217172c9ad6ea0101db251466000000005c15
```
Can you make any sense of it?

Hypfer, 2024-04-09:
Oh interesting. I was wondering if it would work with the new SUN-Mxx inverters or if they've changed something about the protocol. Turns out it's the latter.
I was kinda expecting it to be this was though after having seen the Deye SUN-12K packets and those being entirely different as well.

The data you're receiving isn't garbage. The parser just doesn't know how to make sense of it.

What is interesting is this:
> LSW5BLE_MW17_5406_RELAY_270_1.00-SH

Is your SUN-M80 one with the NA relay integrated? Afaik there were some units that required the external SUN-MI-RELAY-01 and others that come with it inside it.
If so, this kinda looks a bit like as if the internal relay module to the cloud still pretends to be an external one or something like that.

There are a few ToDos here.

First of all, you should check if the data is actually correct and not just plausible. For that, either compare with what you see in the solarman cloud or check what you can read via modbus.

Then, if the hypothesis is correct, someone unfortunately would have to use the solarman cloud and see if they can see any data from that relay module. If yes, they'd then have to sample the data there while also doing a package capture so that one can make sense of the package structure by comparing values in it with known data at a given point in time.

If the data you see in the first packets is actually correct and there's nothing relevant in the relay packets, it might be enough to figure out a way to detect that they are relay packets and discard them

czoller, 2024-04-10:
Thanks for your answer, @Hypher. It's a bit annyoing that I can only test it during the day. I hope, I can try something out soon.

czoller, 2024-04-10:
> First of all, you should check if the data is actually correct and not just plausible. For that, either compare with what you see in the solarman cloud or check what you can read via modbus.

For me the content of the first packet looks quite correct.

This is what I get here:

```json
{
  "pv": {
    "1": {
      "v": 30.8,
      "i": 1.2,
      "w": 36.96,
      "kWh_today": 0,
      "kWh_total": 82
    },
    "2": {
      "v": 31.2,
      "i": 1.2,
      "w": 37.44,
      "kWh_today": 0,
      "kWh_total": 89.5
    },
    "3": {
      "v": 0,
      "i": 0,
      "w": 0,
      "kWh_today": 0,
      "kWh_total": 0
    },
    "4": {
      "v": 0,
      "i": 0,
      "w": 0,
      "kWh_today": 0,
      "kWh_total": 0
    }
  },
  "grid": {
    "active_power_w": 79,
    "kWh_today": 0,
    "kWh_total": 180.3,
    "v": 239,
    "i": 0.3,
    "hz": 50
  },
  "inverter": {
    "radiator_temp_celsius": 23.7
  },
  "inverter_meta": {
    "rated_power_w": 800,
    "mppt_count": 2,
    "startup_self_check_time": 60,
    "current_time": "2024-04-10T15:38:07.000Z",
    "grid_freq_hz_overfreq_load_reduction_starting_point": 50.2,
    "grid_overfreq_load_reduction_percent": 44,
    "grid_v_limit_upper": 275,
    "grid_v_limit_lower": 183,
    "grid_freq_hz_limit_upper": 51.5,
    "grid_freq_hz_limit_lower": 47.5,
    "protocol_ver": "V0.2.0.1",
    "dc_master_fw_ver": "V0.1.1.7",
    "ac_fw_ver": "V1.2.1.8"
  }
}
```
Compared to what jedie/inverter-connect> returns: 
```
┏━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━┓
┃ Counter ┃                                            Name ┃ Value                    ┃ Register ┃ Length ┃             Raw data ┃
[... 21 lines cut ...]
│      21 │                                   ON-OFF Enable │           ON             │   002B   │   1    │                 0001 │
│      22 │                        Island Protection Enable │      Enabled             │   002E   │   1    │                 0001 │
│      23 │              Overfrequency&Load-shedding Enable │      Enabled             │   0031   │   1    │                 0001 │
│      24 │                                  Running Status │        Fault             │   003B   │   1    │                 0004 │
│      25 │                  Total AC Output Power (Active) │         79.0 W           │   0056   │   2    │             03160000 │
│      26 │                            Radiator Temperature │         23.7 °C          │   005A   │   1    │                 0d2a │
│      27 │                                     Inverter ID │ ********************     │   0003   │   5    │ ******************** │
│      28 │                                Hardware Version │     v0.1.0.2             │   000C   │   1    │                 0102 │
│      29 │                      DC Master Firmware Version │     v0.1.1.7             │   000D   │   1    │                 0117 │
│      30 │                              AC Version. Number │     v1.2.1.8             │   000E   │   1    │                 1218 │
│      31 │                                     Rated Power │        800.0 W           │   0010   │   1    │                 1f40 │
│      32 │                  Communication Protocol Version │     v0.2.0.1             │   0012   │   1    │                 0201 │
│      33 │                     Start-up Self-checking Time │           60 s           │   0015   │   1    │                 003c │
│      34 │                                     Update Time │ <raw hex: 18040a0f2b0b>  │   0016   │   3    │         18040a0f2b0b │
│      35 │                               Soft Start Enable │     Disabled             │   002F   │   1    │                 0000 │
│      36 │                         Power Factor Regulation │         -4.3             │   0032   │   1    │                 ffd5 │
│      37 │                        Restore Factory Settings │     Disabled             │   0036   │   1    │                 0000 │
│      38 │                                       PV1 Power │        36.96 W           │    -     │   -    │           (Computed) │
│      39 │                                       PV2 Power │        37.44 W           │    -     │   -    │           (Computed) │
│      40 │                                     Total Power │         74.4 W           │    -     │   -    │           (Computed) │
└─────────┴─────────────────────────────────────────────────┴──────────────────────────┴──────────┴────────┴──────────────────────┘
```
The only difference is the total production, but here the one from dummycloud looks more reasonable. 

The values also match with those in the web UI.

Hypfer, 2024-04-10:
For debugging, I'd recommend adding some code here that either logs as hex or writes to a file so that you can share one of the packets that fail to parse properly
https://github.com/Hypfer/deye-microinverter-cloud-free/blob/e35914b2d1f08cc1431ccd628f65427f390acc90/dummycloud/src/DummyCloud.js#L31

czoller, 2024-04-10:
Sorry, I looks like I had removed that line by accident for the second packet in my first post. Here is one group of packets with HEX and JSON:

```
[2024-04-10T15:43:10.935Z] [TRACE] 2024-04-10T15:43:10.935Z Data a54a01104231322c9ad6ea0108541d75680049100000723eae650100cf060000323330353130373134360400000000000b0700000000000056090000000003000000000088134f000000420918fc18fc18fc18fc00000000000000000000000034010c0038010c00000000000000000056302e322e302e3156302e312e312e3756312e322e312e38000000001f40020100040000000000000000000003340000037f0000000000000000323330353130373134362d3120202020323330353130373134362d3220202020323330353130373134362d3320202020323330353130373134362d3420202020139c002c00010000000000000abe0726141e128e003c18040a0f26070001000102011712180000000000000000000000000000000000ff000000000000139c002c00000000000000000049000000000001000000000abe0726141e128e7f3d0000000000000000000000020815
[2024-04-10T15:43:10.936Z] [DEBUG] DATA packet data from ::ffff:192.168.179.10 {
  pv: {
    '1': { v: 30.8, i: 1.2, w: 36.96, kWh_today: 0, kWh_total: 82 },
    '2': { v: 31.2, i: 1.2, w: 37.44, kWh_today: 0, kWh_total: 89.5 },
    '3': { v: 0, i: 0, w: 0, kWh_today: 0, kWh_total: 0 },
    '4': { v: 0, i: 0, w: 0, kWh_today: 0, kWh_total: 0 }
  },
  grid: {
    active_power_w: 79,
    kWh_today: 0,
    kWh_total: 180.3,
    v: 239,
    i: 0.3,
    hz: 50
  },
  inverter: { radiator_temp_celsius: 23.7 },
  inverter_meta: {
    rated_power_w: 800,
    mppt_count: 2,
    startup_self_check_time: 60,
    current_time: 2024-04-10T15:38:07.000Z,
    grid_freq_hz_overfreq_load_reduction_starting_point: 50.2,
    grid_overfreq_load_reduction_percent: 44,
    grid_v_limit_upper: 275,
    grid_v_limit_lower: 183,
    grid_freq_hz_limit_upper: 51.5,
    grid_freq_hz_limit_lower: 47.5,
    protocol_ver: 'V0.2.0.1',
    dc_master_fw_ver: 'V0.1.1.7',
    ac_fw_ver: 'V1.2.1.8'
  }
}

[2024-04-10T15:43:11.046Z] [TRACE] 2024-04-10T15:43:11.046Z Data a54a01104232332c9ad6ea810854025c680060210000000000000100b70600003233303531303731343604005401000008070000000000005609000000000700000000008813b4000000640a18fc18fc18fc18fc00000000000000000000000040011a003c011d00000000000000000056302e322e302e3156302e312e312e3756312e322e312e38000000001f4002010004000f001100000000000003330000037d0000000000000000323330353130373134362d3120202020323330353130373134362d3220202020323330353130373134362d3320202020323330353130373134362d3420202020139c002c00010000000000000abe0726141e128e003c3000000000000001000102011712180000000000000000000000000000000000ff000000000000139c002c00000000000000000049000000000001000000000abe0726141e128e7f3d0000000000000000000000029b15
[2024-04-10T15:43:11.046Z] [DEBUG] DATA packet data from ::ffff:192.168.179.10 {
  pv: {
    '1': { v: 32, i: 2.6, w: 83.2, kWh_today: 1.5, kWh_total: 81.9 },
    '2': { v: 31.6, i: 2.9, w: 91.64, kWh_today: 1.7, kWh_total: 89.3 },
    '3': { v: 0, i: 0, w: 0, kWh_today: 0, kWh_total: 0 },
    '4': { v: 0, i: 0, w: 0, kWh_today: 0, kWh_total: 0 }
  },
  grid: {
    active_power_w: 180,
    kWh_today: 3.4,
    kWh_total: 180,
    v: 239,
    i: 0.7,
    hz: 50
  },
  inverter: { radiator_temp_celsius: 26.6 },
  inverter_meta: {
    rated_power_w: 800,
    mppt_count: 2,
    startup_self_check_time: 60,
    current_time: Invalid Date,
    grid_freq_hz_overfreq_load_reduction_starting_point: 50.2,
    grid_overfreq_load_reduction_percent: 44,
    grid_v_limit_upper: 275,
    grid_v_limit_lower: 183,
    grid_freq_hz_limit_upper: 51.5,
    grid_freq_hz_limit_lower: 47.5,
    protocol_ver: 'V0.2.0.1',
    dc_master_fw_ver: 'V0.1.1.7',
    ac_fw_ver: 'V1.2.1.8'
  }
}
[... 316 lines cut ...]
    active_power_w: 0,
    kWh_today: 0.49,
    kWh_total: 89235128.2,
    v: 1387.6,
    i: 0.8,
    hz: 0
  },
  inverter: { radiator_temp_celsius: 0 },
  inverter_meta: {
    rated_power_w: 30,
    mppt_count: 0,
    startup_self_check_time: 0,
    current_time: Invalid Date,
    grid_freq_hz_overfreq_load_reduction_starting_point: 0,
    grid_overfreq_load_reduction_percent: 256,
    grid_v_limit_upper: 0,
    grid_v_limit_lower: 0,
    grid_freq_hz_limit_upper: 0,
    grid_freq_hz_limit_lower: 0,
    protocol_ver: '\x00\x02\x00\x00\x00\x02\x0B;',
    dc_master_fw_ver: '\x0B;\x07&\x04\x0B\x04\x0B',
    ac_fw_ver: '\x14\x1E\x14\x1E\x12\x0E\x12\x0E'
  }
}
```

czoller, 2024-04-10:
For me, the easiest way for now is to just discard those packets, that don't have a proper date, after parsing. But that might not be the proper solution that you have in mind...

Hypfer, 2024-04-10:
Hm. Those data packets that get parsed correct-ish looking have what I called `payload_magic` in the kaitai struct set to `129` as opposee to the expect`1`. Maybe that should be read as `0b10000001` and `0b00000001` respectively and that bit is a flag?

It looks like it's data from a prior point in time.
But why would the inverter send that to the cloud?

Wild guess but maybe to combat that there are holes in their graphs if there was a short internet outage and therefore the logger now keeps n to n-10 packets and always sends them on each report interval?

Hypfer, 2024-04-10:
Please try https://github.com/Hypfer/deye-microinverter-cloud-free/commit/b29b255268695acc0fe1ff0c50704cb4d9470d84

czoller, 2024-04-10:
Thanks a lot. Tomorrow, I'm not at home during the day. I'm gonna try it on Friday.

czoller, 2024-04-12:
> Please try [b29b255](https://github.com/Hypfer/deye-microinverter-cloud-free/commit/b29b255268695acc0fe1ff0c50704cb4d9470d84)

Looks good.

But, is it normal that the timestamps in `inverter_meta.current_time` doesn't correspond to the actual current time but to the time of the previous round?

| Actual Time                           | inverter_meta.current_time |
| -------------------------------------- | ---------------------------------------- |
| 2024-04-12T16:08:31.010Z | 2024-04-12T16:07:56.000Z   |
| 2024-04-12T16:13:34.114Z | 2024-04-12T16:08:31.000Z   |
| 2024-04-12T16:18:38.143Z | 2024-04-12T16:13:34.000Z   |
| 2024-04-12T16:23:41.551Z | 2024-04-12T16:18:38.000Z   |
| 2024-04-12T16:28:49.474Z | 2024-04-12T16:23:42.000Z   |
| 2024-04-12T16:38:53.446Z | 2024-04-12T16:28:49.000Z   |

Hypfer, 2024-04-14:
If you take the first packet from the log you've posted 
> [2024-04-10T15:43:10.935Z] [TRACE] 2024-04-10T15:43:10.935Z Data [...]

and decode that, you'll find that the date contained is `18 04 0a 0f 26 07` which is `24, 4, 10, 15, 38, 7` in decimal meaning `2024-04-10 15:38:07`

So yeah, that's actually what the logger reports. I'd suspect that the timestamp is the **start** of the 5 minute sampling period over which the data is averaged.

czoller, 2024-04-14:
> If you take the first packet from the log you've posted
> 
> > [2024-04-10T15:43:10.935Z] [TRACE] 2024-04-10T15:43:10.935Z Data a54a...
> 
> and decode that, you'll find that the date contained is `18 04 0a 0f 26 07` which is `24, 4, 10, 15, 38, 7` in decimal meaning `2024-04-10 15:38:07`

Yes, I had no doubt that the parser is correct, I'm was just wondering if that's the normal behavior of these inverters.

> So yeah, that's actually what the logger reports. I'd suspect that the timestamp is the **start** of the 5 minute sampling period over which the data is averaged.

Or the inverter has no internal wall clock at all and just stores and passes the last timestamp received from the cloud (which leads to the same effect in the end).

Hypfer, 2024-04-14:
> Or the inverter has no internal wall clock at all and just stores and passes the last timestamp received from the cloud (which leads to the same effect in the end).

Sure that is possible, but what would be the point of doing that?

czoller, 2024-04-14:
Of doing what? Having no internal wall clock or storing and passing the last timestamp?

closed 2024-04-14 by Hypfer

czoller, 2024-04-14:
@Hypfer : Thanks a lot for your help and your whole work on this project!