Hypfer/deye-microinverter-cloud-free #21: Inverter Connecting but producing error, no activity on MQTT side

Xnyle, 2024-12-06:
From the logs
[2024-12-06T13:17:48.333Z] [INFO] Attempting to reconnect to MQTT broker
[2024-12-06T13:17:51.405Z] [INFO] Attempting to reconnect to MQTT broker
[2024-12-06T13:17:54.481Z] [INFO] Attempting to reconnect to MQTT broker
[2024-12-06T13:17:54.531Z] [INFO] Connected to MQTT broker
[2024-12-06T13:59:57.982Z] [INFO] New connection from ::ffff:192.168.66.212
[2024-12-06T13:59:58.000Z] [ERROR] Error while parsing packet from ::ffff:192.168.66.212 Error: Invalid header magic: 22
    at Protocol.parseHeader (/app/src/Protocol.js:20:19)
    at Protocol.parsePacket (/app/src/Protocol.js:56:33)
    at Socket.<anonymous> (/app/src/DummyCloud.js:35:41)
    at Socket.emit (node:events:524:28)
    at addChunk (node:internal/streams/readable:561:12)
    at readableAddChunkPushByteMode (node:internal/streams/readable:512:3)
    at Readable.push (node:internal/streams/readable:392:5)
    at TCP.onStreamRead (node:internal/stream_base_commons:189:23)

Inverter is Sun800G3
MW3_SSL_5408_1.0B

Xnyle, 2024-12-06:
Trace gives this packet

[2024-12-06T14:22:36.101Z] [TRACE] 2024-12-06T14:22:36.101Z Data ::ffff:192.168.66.212: 160303007f0100007b03032ffc7667ece59481e6fb592412be93db7878e635340d8ab44ac65ab152d8f92f000036006bc00ac0140039c02bc02f009ec023c0270067c009c0130033003d0035c00fc005009c003c002fc031c029c00ec02dc025c00400ff0100001c000a000400020017000d0006000404030401000b0002010000230000

Xnyle, 2024-12-07:
OK, it's because of new SSL firmware. MW3_SSL_5408_1.0B

I tried to downgrade to 2.06 and also 1.57, i can upload the new FW, but it seems inverter refusing to install old FW on MW3_SSL_5408_1.0B.

Has anyone been able to downgrade from that version?

m0d3rnX, 2025-03-09:
I have the same problem, i restricted the internet connectivity too late. 
Other people are saying that you can write the support at [servicede@deye.com.cn](mailto:servicede@deye.com.cn) (it does seem german, you can try it in english too)

I will edit this if it worked for me.

Hypfer, 2025-03-09:
@Xnyle Do you have more logs?

I'd be curious how much the packets change or don't change