Hypfer/deye-microinverter-cloud-free #23: Hybrid inverter packet

judedaryl, 2025-01-14:
How were you able to figure out how to parse the payload packet for microinverter? I have a hybrid inverter and would like to create my own personal remote server for it.

Hypfer, 2025-01-14:
Essentially, you need to know which metrics were at which values at some point in time and then try to find matching bytes within the packet. Do that a bunch of times to eliminate false matches and you eventually find the right bytes.

Thus, what you'd need to do is sample via modbus at the same time as capturing cloud traffic. Or capturing cloud traffic and at the same time scraping the cloud webinterface.
Then, all you need is time and staring at the captures until it starts making sense

I'm using ide.kaitai.io to document the format.  The struct can be found here:
https://github.com/Hypfer/deye-microinverter-cloud-free/blob/master/assets/deye_sun600.ksy

There's some work on hybrid inverter packets already in there, though it would be a good idea to double-check that, because I only had pcaps of one hybrid inverter.

What is also still completely missing is commands sent via the cloud so if your inverter is cloud connected, maybe use that to create some pcaps while triggering actions/changing settings via the cloud.

Let me know if you figure something out.
Happy hacking!

judedaryl, 2025-01-15:
thanks @Hypfer ! let me look into it.